vortigont/FireLamp_JeeUI #23: Случайный эффект в ДЕМО

dina16121974, 2023-06-22:
Фит 32. Сбоят некоторые эффекты в этом режиме. Конкретно эффект 24 прыгающие мячики ,опять с краю матрицы прыгают. Буду добавлять ,сюда замеченное.

dina16121974, 2023-06-22:
Эффект 72 лавовая лампа ,иногда с косяками запускаеться ,видел не горящие ряд пикселей сверху и справа.

vortigont, 2023-06-22:
в этом режиме все контролы дергаются случайным образом, там может что угодно выстрелить.
Если заметили некрасивость, то описывайте максимально точно, т.к. просто так понять при каких настройках это воспроизвести - невозможно. Число мячиков, форма, положение все контролов и т.п.
И лучше не сваливать проблемы разных эффектов в одну заявку, т.к. закрыть и отслеживать её будет сложно

dina16121974, 2023-06-24:
Тут не в контролах дело ,а именно на старте ,не соответствие картинки и установленных контролов. Допустим мячики стоит 3 ,а показывает больше. Пока в этих 2 заметил. А лавовая лампа ,вообще может перезагрузку вызывать ,если контролы крутишь.

vortigont, 2023-06-24:
надо смотреть оно так задумано или нет - вероятно состояние контролов определяется сохраненными данными, а не тем что рандомно выбрало демо
по "лавовой лампе" отдельно попробуйте отловить проблему вне демо

vortigont, 2024-05-30:
Not a  bug

closed 2024-05-30 by vortigont