vortigont/FireLamp_JeeUI #35: Хаб 75

dina16121974, 2023-10-19:
Эффекты которые ,не идут. 1 ,10 ,12 ,13 ,15 (половина панели работает) ,20 ,21 ,24 ,25 ,28 ,29 ,33 ,37 ,42 ,42 ,44 ,47 ,49 ,51, 56, 57, 59, 60, 65, 69, 70 (половина панели работает) ,71, 73, 254, 255. И еще чередование цветов ,не то. Надо где то настройку сделать.

vortigont, 2023-10-19:
Поясните что значит (половина панели работает) и не идут. Я же не телепат. 1й эффект это белая заливка, что там может не работать-то? Шкалу крутили?
А чередование цветов меняется перетыканием проводков идущих в панель.

dina16121974, 2023-10-19:
Шкалу крутил. Эффект крутиться на одной половине панели. Не идут ,перезагрузка. Позже логи попробую выложить.

dina16121974, 2023-10-19:
Не у всех эффектов перезагрузка. Многие в дебаге видно выполняються ,но панель не горит.

dina16121974, 2023-10-19:
> Не у всех эффектов перезагрузка. Многие в дебаге видно выполняються ,но панель не горит. Чередование мне паяльником надо перетыкивать,хер с ним 2 панели есть ,на одной нормально. Подожду от Вас когда выбор пинов добавите и может чередование цвета.

dina16121974, 2023-10-27:
> Эффекты которые ,не идут. 1 ,10 ,12 ,15 (половина панели работает) ,20 ,21 ,24 ,25 ,28 ,29 ,33 ,37 ,42 ,42 ,44 ,47 ,49 ,51, 56, 57, 59, 60, 65, 69, 70 (половина панели работает) ,71, 73, 254, 255. И еще чередование цветов ,не то. Надо где то настройку сделать.

dina16121974, 2023-11-23:
Закрыть можно.

vortigont, 2023-11-24:
настройка реализована в 04d081a

closed 2023-11-24 by vortigont