vortigont/FireLamp_JeeUI #37: MQTT Home Assistant

BashSer, 2023-10-30:
Приветствую, @vortigont !

> Немножко нытья. Не так давно купил **готовый комплект**, думал всё будет удобно и приятно, начну знакомится с esp* и подобным, но всё оказалось плохо. Инструкции нет, просто рассыпуха с другим заказом, ничего не понятно, нашёл автора AlexGyver. Чем дальше я искал, тем всё хуже становилось... Множество прошивок, где-то политику приплели, где-то целые форки снесли, никаких ответов, однотипный спам вопросов на сотни страниц и т.д. Конец.

Я понял следующее:
- Начало проекта тянется из 2018-19 года. Изначально написан AlexGyver в Arduino IDE, но по-своему наспех и потом не особо поддерживался. Остановилось на версии 1.4;
- Дальнейшее развитие проекта разделилось на несколько "программных веток" от нескольких авторов. Из множества, что у меня сразу заработало без какого-либо шаманства и к тому же сразу появилось в HA(Home Assistant) по MQTT -> версия 1.6 MQTT https://github.com/Whilser/GyverLamp/tree/master/firmware . Минусы - да, это всё тоже самое продолжение старого хаоса;
- Сейчас снова наблюдается активность и Ваша активность похоже самая лучшая. В хорошем смысле завидую Вашему упорству в разборке проектов. Вы большой молодец.
- На форуме Вы просили писать сюда - ну что ж, вводное сообщение закончилось - к делу ! =)
- upd: благодаря Вам перешёл тоже на Platform IO, по шпаргалке могу компилировать и прошивать. Просто установил и ещё не разбирался

Прошивка 1.6 MQTT https://github.com/Whilser/GyverLamp/tree/master/firmware
При первом запуске esp8266(пока что, esp32 заказаны) поднимается точка доступа, подключаемся к ней и заходим на шлюз, выбираем в меню первый пункт настройки Wi-FI видим:
[image]
(Первый пункт меню настройки Wi-Fi. Тут же добавили настройки для MQTT)

Перезагружаем esp8266, в консоли видно как она использует ранее введённые данные, подключается к роутеру и к MQTT брокеру в HA. Лампа сама появляется в MQTT и просит настройки, просто присваиваем комнату и готово:
[image]

Дальше скриншоты как она видна в HA и что может сделать:
[image]
(базовая настройки карточки на главном столе. HA быстро развивается и имеет огромное дружелюбное коммунити. У них был год голосовых помощников, год видеокамер и т.д. Кое-что тут новое для меня, пару лет назад было слегка иначе, я помогал в настройке нескольких квартир и теперь делаю родителям и себе)

[image]
(При нажатии на карточку открывается такое окно. Можно вкл\выкл, яркость в процентах, выбор эффектов и просто цвет из палитры на соседней вкладке)

Также она сама везде прописывается и принимает нужные параметры. В дальнейшем их можно использовать для автоматизаций и обращения через MQTT. Многие используют Node-Red для построения автоматизация - там же можно рассылать нужные параметры. Прикладываю диагностические данные с устройства: [config_entry-mqtt-823269d87d8fe113ea4ee0e4d2a1bb03.json.txt](https://github.com/vortigont/FireLamp_JeeUI/files/13209808/config_entry-mqtt-823269d87d8fe113ea4ee0e4d2a1bb03.json.txt)
На что обратить внимание ? Ctrl+F -> "homeassistant/light/ESP-27c40a/"
Через MQTT будут отсылатся сообщения с разным Payload в topic homeassistant/light/ESP-27c40a/что-то там
Вот небольшой кусочек, какие параметры есть в данном случае(оставил комментарии):
```
"discovery_data": {
            "topic": "homeassistant/light/ESP-27c40a/config",
            "payload": {
              "name": "Gyver Lamp ESP-27c40a",
              "unique_id": "27c40a",
              "availability_topic": "homeassistant/light/ESP-27c40a/state",
              "payload_available": "online",
              "payload_not_available": "offline",
              "brightness_command_topic": "homeassistant/light/ESP-27c40a/brightness/set", # brightness/set - установить яркость
              "brightness_state_topic": "homeassistant/light/ESP-27c40a/brightness/status", # brightness/status - текущая яркость
              "brightness_scale": 255,
              "command_topic": "homeassistant/light/ESP-27c40a/switch", # вкл\выкл
              "state_topic": "homeassistant/light/ESP-27c40a/status", # сейчас вкл\выкл статус
              "effect_command_topic": "homeassistant/light/ESP-27c40a/effect/set", # Выбрать эффект
              "effect_state_topic": "homeassistant/light/ESP-27c40a/effect/status", # Какой эффект сейчас выбран
              "rgb_command_topic": "homeassistant/light/ESP-27c40a/rgb/set", # Установить цвет RGB
              "rgb_state_topic": "homeassistant/light/ESP-27c40a/rgb/status", # Какой цвет RGB сейчас залит на лампе
              "device": {
                "identifiers": [
                  "27c40a" # Применительно к такому-то устройству
                ], #Может быть версия ниже чем-то поможет, я не знаю
                "name": "Gyver Lamp",
                "manufacturer": "Alex Gyver",
                "model": "Gyver Lamp v2",
                "sw_version": "1.5.5 MQTT"
```
(так же в ранее упомянутом репозитории, а именно вот тут https://github.com/Whilser/GyverLamp/blob/master/firmware/GyverLamp_v1.6_MQTT/mqtt.ino - есть логика и по каким параметрам что-либо управляется)

- Что хотелось бы ?
Ввести минимальную поддержку MQTT из прошивки выше. С первого взгляда это хорошее начало для MQTT.

- Какие возможности открываются и что можно добавить ? (в поддержку введения доступного MQTT)
Даже базовые вкл\выкл, яркость и RGB дают возможность управлять и автоматизировать лампу. Можно зациклить таймер в Node-Red и вот будильник-рассвет готов. В зависимости от дня\погоды\датчика\придумайте сами менять цвет, пусть хоть мигалкой моргает при звонке или получении уведомления\статуса\сообщения, что опять же можно прицепить и как обнаружение хозяина и приветствовать его, хоть пусть красным загорается когда кто-то неудобно пароль ввел. В общем, всё зависит от полёта фантазии.

- Чем я готов помочь ?
Из-за большой нагрузки на работе не успеваю в своих проектах. Сейчас потихоньку разбираюсь с докером и теперь через контейнеры поднимаю два HA на Raspberry Pi 4. По вечерам могу лить прошивки на esp8266(пока что, я понимаю и принимаю переход на esp32, но они ещё едут) и рассказывать о результате. После приезда и настройки камеры могу дать доступ в тестовый HA(можно взорвать, не страшно) - можно смотреть за результатом благодаря камере и тыкать разные кнопки из веб-интерфейса 24\7. За одно зайти в события MQTT и посмотреть как оно выглядит.
Ну и само собой высказывать хотелки через Issue, тестировать и хотелось бы подключиться к разработке, но пока не хватает знаний и времени на это. Внезапно, я осознал как эта лампа интересна разным группам пользователям и поэтому при предложении чего-либо это будет из разряда "хотелось фичу, потому что взрослым так, детям так и она покрывает сразу несколько хотелок", а не "хочу фичу, потому что только мне нравится, о других не думаю и пофиг что там очень сложно".
Прошу извинения за огромную кучу текста, пока хочу понять как с Вами общаться и какая именно информация Вас интересует.

vortigont, 2023-10-31:
Добрый!
Спасибо за доброе слово. Проект поддерживается и развивается по наличию свободного времени - то пусто, то густо.
Сама рабочая лампа для меня не самоцель, я работаю толковой над архитектурой по мере возможности, поэтому в проекте всё еще не хватает многих вещей. Если нужен готовый проект, гляньте [этот](https://github.com/vvip-68/LedPanelWiFi). Я сам не пользовался, но он довольно неплохо документирован, что уже большой плюс само по себе.
По поводу HA будет здорово если сможете подключиться к адаптации и заняться написанием шаблонов и тестированием. Я с ХА толком не работал и с первого взгляда он мне не понравился - слишком большой порог входа и много возни с программированием. Т.е. это надо сесть и отдельно заниматься его изучением, у меня на это сейчас ни нужды ни времени нет. Но! Я сейчас работаю над унифицированным АПИ для проекта лампы, когда закончу каких-то частичных поддержек не будет - будет бесшовная интеграция UI через вебсокет, мкутт и рест апи. Формат данных будет везде одинаковый через сериализованный джейсон. Так что через мкутт можно будет выполнять все те же функции что из через вебморду без необходимости нарезать какие-то отдельные топики и пр.
Нужно будет это обернуть шаблонами для ХА (если возможно), которыми он сможет разбирать/формировать соответсвующие джейсон объекты для управления.
Как раз пока ваши есп32 доедут, может что-то пробное выкачу.
А так поболтать и обсудить общие вопросы заходите на форум гайвера (можно гитхабовской учеткой авторизоваться).

BashSer, 2023-11-01:
Благодарю за наводку! Буду сегодня пробовать старую версию, т.к. в новой убрали поддержку MQTT.

С радостью, по возможности, постараюсь подключиться. Мне интересен именно HA и MQTT + ESP*, т.к. что действительно почему бы не попробовать. Простой zigbee датчик температуры с aliexpress будет стоить порядка тысячи рублей. На одной даже esp8266 можно не только датчик температуры поместить, а ещё и реле, и кнопки, т.е. выключатели и прочее, и всё это вместе обойдётся в два раза дешевле. Не говоря о том, что ещё можно будет под себя всё настроить и не бояться блокировок в сторонние облака.
Я заинтересовался HA где-то с версии 2020.12 - развитие невероятно бурное. БУКВАЛЬНО полугодовая документация, статьи и видео совершенно не подходят. Тогда и в правду был большой порог входа и очень много ковырять в YAML конфигах. Сейчас все хотелки юзеров добавили и соответственно всё уже везде настроено - ничего не надо добавлять, я по привычке сразу поставил файл-менеджер, а он мне даже до сих пор не пригодился сейчас. Любое устройство в сети само определяется и предлагается добавить будь то включенный телевизор, компьютер, локальные серверы, снмп к роутеру, смартфоны и всего подряд. Для экстремальных любителей кодить можно поставить соответствующий дашборд на питоне. В несколько кликов можно создать список покупок и включить напоминалку, что когда твой телефон(пользователь, GPS) будет проходить мимо магазина отправить напоминалку с нужными штуками. Отдельные дашборды можно применить к разным пользователям - очень удобно, например для тех же родителей будет упрощённый вариант с большими кнопками, а если ребёнок, то показывать свои кнопки для мультиков и без управления светом или ещё чего-нибудь. Что-то я затянул... В общем, по своему опыту каждый год он становился дружелюбнее к новым пользователям - в данный момент это другая система чем год назад и уж тем более ещё раньше. Уже "обучил" несколько человек с разными уровнями знаний самостоятельно устанавливать и админить сие добро.

Про API интересно - опять же с радостью постараюсь подхватить по части HA и MQTT. К слову, на ранее упомянутом Вами репозиторию нашёл в wiki раздел про API, но Вы скорее всего и так уже знаете о нём :)
За приглашение на форум благодарю, но я там больше редким зрителем буду подсматривать :)

Последний вопрос перед закрытием. Сейчас ожидается крупный апдейт и создавать новые Issue не имеет смысла ? Например, в режиме "рисования" в веб-браузере на LED матрице картинка отображается повёрнутой на 90 градусов - какие бы параметры подключения матрицы в коде не менял, всё без изменений. В браузере закрашиваешь фигуру в виду "П", а на матрице отображается "П" только упавшая набок.
Или вопрос по веткам есть - в readme написано, что теперь для esp8266 будет ветка "legacy", а сейчас "legacy" более старая чем "compat_8266" и не очень понятно :)

vortigont, 2023-11-02:
> Сейчас` ожидается крупный апдейт и создавать новые Issue не имеет смысла ? 

имеет, если проблема относится к коду из мастер ветки. Если это что-то от 8266 то я забросил эту платформу и исправлять и обновлять её не планирую. Она устарела настолько что ею надо заниматься отдельно, а у меня нет ни времени ни желания на это. Ветки оставил на случай если вдруг кто-то когда-то захочет раскопать эту стюардессу.

> в readme написано, что теперь для esp8266 будет ветка "legacy", а сейчас "legacy" более старая чем "compat_8266"

лучше сразу забудьте про все ветки с 8266. Код в этих ветках и то что есть в актуальных релизах и мастере это уже 2 разных проекта. В `legacy` код, который более-менее стабильно работал на 8266 по отзывам пользователей. Это наследие старого украинского проекта, который я постарался довести до ума что бы он хотя бы не падал.
В `compat_8266` это последний код, который всё ещё собирается под 8266. Да он новее, и там я внес намного больше изменений, но что-то уже может не работать из функционала по сравнению с тем что есть в легаси.

vortigont, 2023-11-12:
@BashSer черновик нового [Device-API](https://github.com/vortigont/FireLamp_JeeUI/wiki/Device-API)
можете потихоньку прикидывать как на это натянуть ХА

BashSer, 2024-01-04:
Приветствую @vortigont  !

Не с первой попытки, но всё же до меня дошли правильные ESP и ещё той проблемой оказалось прошить их через Platformio. Что бы просто прошить - необходимо в открытом мониторе терминала нажать КНОПКУ загрузить прошивку, иначе ошибка типа "COM порт не подключён", а вот что бы загрузить ФС уже вышло как-то случайно и я до сих пор не знаю как получилось(там же надо команду писать и соответствующей кнопки в интерфейсе нет) - нервы не выдержали, пока одной ESP хватит. С кнопкой перезагрузки ESP тоже что-то странное творится.
В итоге получилась одна ESP32-Vroom-32D, которая обзавелась микрофоном MAX9814 и сенсорной кнопкой TP223. Отмечу, что у некоторых эффектов есть переключатель "Микрофон" и тогда эффект по разному реагирует на звуки - забавно, но не у всех эффектов есть такой переключатель. Калибровка микрофона "зависает", но уже есть тема #19 по этому поводу. Со многими компонентами помогает вкл\выкл и тогда исправно начинают работать(кнопка, микрофон, mqtt).

По [документации HA](https://www.home-assistant.io/integrations/light.mqtt#json-payload), [Вашему API](https://github.com/vortigont/FireLamp_JeeUI/wiki/Device-API#mqtt) и пару часов получилось следующее.
Добавить в configuration.yaml HA следующие строки:
```
mqtt:
  - light:
      schema: template
      effect_list:
        - 75
        - 76
      command_topic: "EmbUI/08D1F93D787C/post" #подставить MAC-адрес, если префикс не был указан в настройках ESP
      command_on_template: >
        { "action": "dev_pwrswitch", "data": {"dev_pwrswitch": true}
        , "action": "eff_sw_idx", "data": { "eff_sw_idx": {{effect}} }
        }
      command_off_template: '{ "action": "dev_pwrswitch", "data": {"dev_pwrswitch": false}}'
```
После перезагрузки конфига добавить на панель новый объект и получается следующая карточка:
[image]

Полностью или частично отваливается MQTT со стороны ESP - либо перестаёт что-то говорить в топики, либо только последнюю команду принимает, помогает перезагрузка ESP по питанию, перезагрузка кнопками на ESP и из веб-интерфейса не помогает. Иногда веб-интерфейс не отвечает на ESP или самопроизвольно перезагружается - причину понять не могу.

Буду дальше пробовать, согласно документации можно распарсить список эффектов при помощи effect_template, да и в общем сделать небольшой шаблон со всеми if и определять какая матрица, доп. устройства и т.д.

Ещё надо поизучать интеграцию [WLED](https://www.home-assistant.io/integrations/wled/) - возможно, там будет всё намного проще. Или [ESPHome](https://esphome.io/components/light/index.html) - подключил ESP, нажал прошить(само поднимает Platformio внутри HA) и потом пару строк что это и какой пин занимает, так же упоминается WLED в документации. Если там будет поддержка WLED, то без проблем можно будет всю оснастку подружить, включая датчики, реле, и т.д.

vortigont, 2024-01-04:
Привет.
Нормальное начало с ХА!
```
{ "action": "dev_pwrswitch", "data": {"dev_pwrswitch": true} ,
 "action": "eff_sw_idx", "data": { "eff_sw_idx": {{effect}} }
 }
```
это невалидный джейсон, здесь 2 ключа с одним именем. Разбейте команду на две или вообще опустите эффект, будет включен последний рабочий.

Насчет WLED и ESPHOME это отдельные сущности. Под влед шаблоны и джейсон апи написать можно, я уже делал это под старую версию апликухи. Но особого смысла в этом не вижу, т.к. влед использует тотже самый вебвью и открывает браузер для управления, т.е. тоже самое что и сейчас есть. Единственное что это даст - видимость ламп в приложеньке.

ESPHOME это отдельный проект и никаким боком его к моей прошивке не присобачить, у них свои библиотеки, свои манифесты, это свой конструктор.

BashSer, 2024-01-04:
Да, понимаю, что невалидный, но так работает переключение эффектов, а не просто вкл\выкл =)
Но тут есть проблемы, что я не могу указать разные топики, по идее это выглядело бы как-то так:
```
mqtt:
  - light:
      schema: template
      effect_list: [список эффектов]
      command_topic: "EmbUI/08D1F93D787C/set" # топик для установки команд
      state_topic: "EmbUI/08D1F93D787C/status" # топик со статусом
      availability_topic: "EmbUI/08D1F93D787C/online" #топик с доступностью, сообщения о появлении в сети
      command_on_template: > #формирование команды при включении или смены эффекта
        {"state": "on"
        {%- if brightness is defined -%}
        , "brightness": {{ brightness }}
        {%- endif -%}
        {%- if effect is defined -%}
        , "effect": "{{ effect }}"
        {%- endif -%}
        #перечисление атрибутов, если они имеются
        # получится что-то по типу {"state": "on", "brightess": 50...}
        }
      command_off_template: '{"state": "off"}'
      state_template: '{{ value.split(',')[0] }}'  # здесь и далее можно по разному парсить ответы для формирования команды выше
      brightness_template: '{{ (100 / value_json.brightness | float ) | round(0) }}'
      effect_template: '{{ value_json.effect }}' #можно забрать список эффектов или возможно как-то спарсить контролы при помощи json_attributes_topic + json_attributes_template
```
В таком случае, можно было бы отправлять готовый джейсон со всеми индивидуальными параметрами, но нужно дальше разбираться с этим, а знаний пока мало. Предположительно, в ESP можно указать что-то для Auto-Discovery и тогда всё снова вернётся к небольшому конфигу где бы парсились доступные атрибуты и не придётся что-то менять со временем.
Кстати, по умолчанию ESP32 с релизом 3.4.0 обнаружилась с этими параметрами в HA:
[image]

Ещё вариант сделать файлик автоматизаций на все действия и призывать встроенную службу MQTT Publish для отправки каждого действия, но получится громоздко и потом править неудобно, например:
```
input_number:
  neopixel_animation_speed:
    name: NeoPixel Animation Speed
    initial: 200
    min: 0
    max: 255
    step: 5

automation:
  - id: "71938579813759813757"
    alias: NeoPixel Animation Speed Send
    initial_state: true
    hide_entity: false
    trigger:
    - entity_id: input_number.neopixel_animation_speed
      platform: state
    action:
    - data_template:
        payload_template: '{"speed": {{ trigger.to_state.state | int }}}'
        retain: true
        topic: home/McLighting01_ha/state/in
      service: mqtt.publish

  - id: "93786598732698756967"
    alias: NeoPixel Animation Speed Receive
    trigger:
    - platform: mqtt
      topic: home/McLighting01_ha/state/out
    action:
    - data_template:
        entity_id: input_number.neopixel_animation_speed
        value: '{{ trigger.payload_json.speed | int }}'
      service: input_number.set_value
```

Теперь хотя бы есть с чем экспериментировать =)

vortigont, 2024-01-04:
> но так работает переключение эффектов, а не просто вкл\выкл =)

ну это результат негарантированный :) Вообще раньше так и было - переключение эффекта при выключенной лапме принудительно её включало. Но потом от этого отказался, есть случаи когда это не нужно.

> Но тут есть проблемы, что я не могу указать разные топики, по идее это выглядело бы как-то так:

так и есть же отдельные - в вики описаны
```
~/get/#
~/set/#
~/post
```
и пр.
Отдельно на каждый чих топика нет конечно ибо это очень муторно и вносит избыточность.

инфо об он-лайн и пр тех информация публикуется в sys топик
[image]

Списки эффектов и контролов я еще не делал, эту инфу морда дергает по http с ФС контроллера, но впринципе добавить можно, обсуждаемо. Как только будет более менее понятно чего хочется получить.

> Кстати, по умолчанию ESP32 с релизом 3.4.0 обнаружилась с этими параметрами в HA:

это артефакты, информация неактуальная. Надо заменить на нормальный шаблон, вот жду того кто его напишет :)

> Ещё вариант сделать файлик автоматизаций на все действия и призывать встроенную службу MQTT Publish для отправки каждого действия, но получится громоздко и потом править неудобно

я не оч понял как это работает )
В общем смысл такой - текущий апи повторяет то что делает веб-морда. Вы можете открыть девконсоль браузера и посмотреть какие джейсоны бегают по вебсокету. Это же можно плевать и в мкутт топики.

BashSer, 2024-01-04:
С топиками я имел в виду, что единственная точка входа только ~/post получается, ибо на "#" ругается:
[image]
(Может быть потому что # является символом комментария в yaml и его никак нельзя экранировать. Пока что не встречал, что бы хэш кто-то использовал)
А в конфиге HA указывается только одна точка входа для команд, в противном случае надо на костылях в автоматизациях HA наплодить много сущностей и кнопок, что бы каждая через встроенную службу отсылала на каждый топик уникальные значения =(

Если убрать хэш, то наверное можно было бы так управлять:
[image]

UPD: Через два часа веб-интерфейс и через приложение native WLED перестают переключаться вкладки, но при этом работает вкл\выкл на текущей странцие. После перезагрузки по питанию всё работает как обычно, но что бы начал работать MQTT надо зайти в настройки и тогда появится первое сообщение от MQTT и всё снова работает. С кнопкой такая же беда - перестаёт работать, после перезагрузки по питанию нужно зайти в настройки кнопки и просто сделать выкл\вкл, либо пересохранить нажатия. С микрофоном нужно зайти в настройки, включить калибровку и сделать выкл\вкл в настройках веб-интерфейса.

vortigont, 2024-01-05:
> С топиками я имел в виду, что единственная точка входа только ~/post получается, ибо на "#" ругается:

не единственная. Я в вики расписал
```
Например запрос на включение лампы можно отправить следующими способами:

    topic: ~/set/dev/pwrswitch, body: {"data":{"dev_pwrswitch":true}}
    topic: ~/post, body: { "action":"dev_pwrswitch", "data":{"dev_pwrswitch":true}}
````
`~/get/#` и `~/set/#` это вайлдкарды, в мкутт `#` означает "любой"

> А в конфиге HA указывается только одна точка входа для команд, в противном случае надо на костылях в автоматизациях HA наплодить много сущностей и кнопок

так надо разные топики или все-таки один? В любом случае доступен и тот и тот вариант, я специально так сделал что `action` ы имеют синтаксис с разделителями `_` по  которым можно обратиться к уникальному топику при желании. Как удобнее будет сделать взаимодействие так и пробуйте, я ХА не занимался плотно никогда, для меня это непаханное поле.

> UPD: Через два часа веб-интерфейс и через приложение native WLED перестают переключаться вкладки, но при этом работает вкл\выкл на текущей странцие

предлагаю эту проблему обсудить в отдельном топике

> > 
> 
> В каком месте поискать хотя бы эти артефакты, что бы хоть от чего-то оттолкнуться =)

игнорируйте то что там есть. Нужно на основе АПИ составить жейсон или хмл шаблон который будет публиковаться лампой при регистрации в мкутт.

т.е. напр для включения лампы использовать
```
topic: ~/set/dev/pwrswitch, body: {"data":{"dev_pwrswitch":true}}
```

для яркости
```
topic: ~/set/dev/brightness body: {"data":{"dev_brightness":<value>, "nofade":true}}
```
текущую яркость ловить в топике:
`~/pub/value`
формат ответа `[{"dev_brightness":0},{"dev_brtscale":20},{"dev_lcurve":2}]`

и т.д. по документации в вики. Покопайтесь, буду отвечать на все вопросы, что надо будет, может допишу или поменяю если не фундаментальные требования будут.

vortigont, 2024-01-05:
Не относится к делу:
> Что бы начал работать MQTT надо зайти в настройки и тогда появится первое сообщение от MQTT и всё снова работает.

Сообщения посылаются при смене страниц или каких-то активностях. При подключении к мкутт публикуется системная информация в топик ~/sys/

> С кнопкой такая же беда - перестаёт работать, после перезагрузки по питанию нужно зайти в настройки кнопки и 
просто сделать выкл\вкл, либо пересохранить нажатия.

"Кнопку" я сейчас переписываю целиком, пока не буду смотреть что там.

> С микрофоном нужно зайти в настройки, включить калибровку и сделать выкл\вкл в настройках веб-интерфейса.

эту кнопку калибровки я уберу, она нефункциональна. Флажёк микрофона остается включеный после перезагрузки в настройках?

Лучше не валите разные проблемы в одну тему, это сложно очень отслеживать.

BashSer, 2024-01-05:
> не единственная. Я в вики расписал
>
> ```
> Например запрос на включение лампы можно отправить следующими способами:
>
>     topic: ~/set/dev/pwrswitch, body: {"data":{"dev_pwrswitch":true}}
>     topic: [...]
 
Я приложил скриншот ошибки в HA(не удалось вызвать службу публикации MQTT. Вайлдкарты запрещены). Просто отправка в ```~/set/``` или ```~/set``` значения  ```{ "data": {"dev_pwrswitch": false}```(или +action) не запускает команду на стороне лампы, но отображается в подписке ```~/pub/value```
Т.е. что бы слать каждую команду в каждый топик(```~/set/dev/pwrswitch```, ```~/set/dev/brightness```, ```~/set/dev/lcurve```) нужны костыли, а концепт заключается в том, что есть одна точка входа и дальше формируется один джейсон для этого входа. Например точка входа ```~/set``` и сообщение вида ```{state: ON, если выбрана яркость, то добавляем brightness: 42, иначе ничего не добавляем в payload, если есть атрибуты, которые обязательны, но они не выбраны, то из шаблона общего шаблона джейсона парсим и забираем нужный параметр и т.д.}```
Прошу Вашей помощи - как корректно отправить значение на ```~/set/``` одним джейсоном с вкл\выкл+номер эффекта ?

> так надо разные топики или все-таки один? В любом случае доступен и тот и тот вариант, я специально так сделал что `action` ы имеют синтаксис с разделителями `_` по которым можно обратиться к [...]

По небольшому опыту в Zigbee2Mqtt + документация HA + Node-RED и нескольких конфигураций HA для друзей, должно быть примерно так:
Подписка на топики:
1. ```state_topic``` вида ```~/status``` - для отслеживания статуса, сюда устройство публикует свои изменения. Например, выключатель физически нажали или датчик движения активировался, что в свою очередь позволяет менять статус в HA и как следствие активировать разные сценарии автоматизации
2. ```availability_topic``` вида ```~/online``` - сюда устройство посылает метрики о доступности, чаще всего китайцы и Tuya ограничиваются простым обнаружением ```payload: true/false```, но по идее там находится IP/MAC/Hostname/SSID/RSSI/etc
3. ```json_attributes_topic```(?) - общий корень вида ```~/``` где публикуется время от времени общий джейсон с основными атрибутами для обновления инфы ```{"batt": 100, "state_l1": on, "temp": 20....}``` и потом разбивается на аттрибуты при помощи ```json_attributes_template```

Публикация команд в топик:
1. command_topic вида ```~/set``` - куда посылать команды на управление. Через эту точку формируется общее сообщение с параметрами по типу ```{"state_center":"OFF","state_left":"OFF","state_right":"OFF"}```

Так же есть payload в топике ```~/config``` для Auto Discovery следующего вида:
```
availability:
  - topic: zigbee2mqtt/bridge/state
    value_template: '{{ value_json.state }}'
  - topic: zigbee2mqtt/Smallroom/availability
    value_template: '{{ value_json.state }}'
availability_mode: all
command_topic: zigbee2mqtt/Smallroom/l1/set
device:
  identifiers:
    - zigbee2mqtt_0xa4c138c4c64bede8
  manufacturer: Lonsonho
  model: 2 gang switch with backlight (X702A)
  name: Smallroom
  via_device: zigbee2mqtt_bridge_0x00124b0024c16d7d
name: L1
object_id: smallroom_l1
origin:
  name: Zigbee2MQTT
  sw_version: 1.35.0
  support_url: https://www.zigbee2mqtt.io
payload_off: 'OFF'
payload_on: 'ON'
state_topic: zigbee2mqtt/Smallroom
unique_id: 0xa4c138c4c64bede8_switch_l1_zigbee2mqtt
value_template: '{{ value_json.state_l1 }}'
platform: mqtt
```

Получается логика такая:
1. Через ```~/config``` получаем параметры для Auto Discovery - что это за устройство, какие топики вложены, что умеет и получаем автоматически собранный объект в HA со всеми выключателями и метриками
2. Через топик для команд ```~/set``` будем слать команды на управление. Если в одном устройстве несколько модулей, то они описаны как ```~/switch_1/set```, ```~/led/set```, ```~/button/set```, ```~/mic/set```, что позволяет их разбить на под-объекты и иметь самостоятельные сущности для управления
3. Для синхронизации состояния устройства, когда им управляют вне HA(нажатие кнопки, управление из другого приложения) слушаем топик ```~/status```
4. Напоследок слушаем ```~/online``` как "пинг" устройства, что оно либо пропало, либо появилось
5. Корень ```~/``` применим скорее всего для протокола zigbee как редкие обновления основных параметров, там payload вида:
```
indicator_mode: null
last_seen: '2024-01-05T12:22:14+03:00'
linkquality: 47
power_on_behavior_l1: null
power_on_behavior_l2: null
state_l1: 'OFF'
state_l2: 'OFF'
```

> > > 
[... 31 lines cut ...]
    {
        "mqtt_user": "mqtt_login"
    },
    {
        "mqtt_pass": "mqtt_password"
    },
    {
        "mqtt_topic": null
    }
]
```

>>С микрофоном нужно зайти в настройки, включить калибровку и сделать выкл\вкл в настройках веб-интерфейса.
>
>эту кнопку калибровки я уберу, она нефункциональна. Флажёк микрофона остается включеный после перезагрузки в настройках?

Да, флажок остаётся активным в настройках микрофона после перезагрузки по питанию. Алгоритм магии, что бы всё работало(не знаю как оно в коде работает, описываю действия после которых становится "хорошо"):
1. Включить питание
2. Зайти в настройки кнопки, сделать выкл\вкл флажком, при необходимости проверить действия щелчков. Если что-то не работает, то зайти в это действие и выйти. Например ```OFF: 1 Click - ON```, редактировать, не изменяя параметров просто нажать обновить
3. Зайти в настройки микрофона, нажать калибровку, переключить флажок выкл\вкл, на экране так и подвиснет калибровка, но эффекты начинают хорошеечно реагировать на микрофон
4. Зайти в настройки MQTT и выйти

>Лучше не валите разные проблемы в одну тему, это сложно очень отслеживать.

Принято, тут все действия уже описаны и добавить нечего, поищу темы куда мигрировать информацию

vortigont, 2024-01-05:
> Я приложил скриншот ошибки в HA(не удалось вызвать службу публикации MQTT. Вайлдкарты запрещены)

не надо слать на решетку #, на вайлдкард слать нельзя, на него можно только подписаться.
`оправлять только блок данных в топики ~/get/# и ~/set/#`
означает что можно отправить сообщенине на `~/set/dev/pwrswitch`  или на `~/set/eff/sw/next` или на `~/get/dev/brightness` или на что угодно еще **вместо** решетки.

> концепт заключается в том, что есть одна точка входа и дальше формируется один джейсон для этого входа

без проблем, эта точка `~/post`, туда и отправляете джейсоны в теле которых указан акшен и параметры к нему.

> и сообщение вида `{state: ON, если выбрана яркость, то добавляем brightness: 42, иначе ничего не добавляем в payload, если есть атрибуты, которые обязательны, но они не выбраны, то из шаблона общего шаблона джейсона парсим и забираем нужный параметр и т.д.}`

так не получится на всё иметь произвольный набор параметров, их в апи десятки (если не за сотню). Разбирать произвольный набор параметров на контроллере будет нереальной задачей, плюс есть еще случаи когда параметры имеют смысл только в группе, указать половину по вкусу не выйдет. Поэтому есть ключ `action` который определяет действие и существует некий опциональный набор параметров к действию

> 1. Через `~/config` получаем параметры для Auto Discovery - что это за устройство, какие топики вложены, что умеет и получаем автоматически собранный объект в HA со всеми выключателями и метриками

это надо написать как раз. Сейчас я не знаю формат того что нужно для автодискавери, собственно это вам задача разработать этот шаблон :))))

> 2\. Через топик для команд `~/set` будем слать команды на управление. Если в одном устройстве несколько модулей, то они описаны как `~/switch_1/set`, `~/led/set`, `~/button/set`, `~/mic/set`, что позволяет их разбить на под-объекты и иметь самостоятельные сущности для управления

я опять вас не пойму, то вы говорите вам одну точку входа, то теперь разные модули. Но в любом случае можно слать и так и так, результат будет один, это сделано для удобства интеграций, т.к. я в этом ничего не понимаю заложил избыточный функционал

> 3\. Для синхронизации состояния устройства, когда им управляют вне HA(нажатие кнопки, управление из другого приложения) слушаем топик `~/status`

для этого есть разные топики, почитайте [вики](https://github.com/vortigont/FireLamp_JeeUI/wiki/Device-API#%D1%81%D0%BF%D0%B8%D1%81%D0%BE%D0%BA-%D1%82%D0%BE%D0%BF%D0%B8%D0%BA%D0%BE%D0%B2-%D0%B8%D1%81%D0%BF%D0%BE%D0%BB%D1%8C%D0%B7%D1%83%D0%B5%D0%BC%D1%8B%D1%85-%D0%B2-%D0%BF%D1%80%D0%BE%D0%B5%D0%BA%D1%82%D0%B5)

есть топик `~/pub/value` для обновление значений контролов, есть топик рефлектор `~/pub/post`

> 4\. Напоследок слушаем `~/online` как "пинг" устройства, что оно либо пропало, либо появилось

есть топик `~/sys/` там вся системная информация о контроллере

> Просто хотел узнать где он находится и как выглядит, может получится хоть в какой-то форме описать этот шаблон

нигде, нет его. Это рудимент от старого кода, нужно написать новый. Если интересно, то смотрите [main.cpp](https://github.com/vortigont/FireLamp_JeeUI/blob/master/src/main.cpp), ищите функцию `String ha_autodiscovery()`
она формирует джейсон, но сейчас он не имеет смысла, оставлен как заглушка.

> При прослушке `~/pub/value` первая информация появляется только после входа во вкладку MQTT

это неверно, в валуе публикуются все **значения** каких либо контролов в вебюи, напр изменение яркости, выбор эффекта и пр. У вас [explorer](http://mqtt-explorer.com/) есть? Я там всё смотрю.

Давайте на примерах:

топик `~/pub/value`

`[{"dev_pwrswitch":false}]` - был выключен чекрыжик питания в вебморде

`[{"dev_brightness":11},{"dev_brtscale":20},{"dev_lcurve":2}]` - изменена яркость на 11/20

`[{"dev_pwrswitch":true}]` - был включен чекрыжик питания в вебморде

`[{"eff_sw_idx":14}]` - был переключен эффект на 14 в выпадающем списке

`[{"eff_sw_next":null}]` - была нажата кнопка "следующий эффект"

топик-рефлектор `~/pub/post`, сюда поступают копии команд от вебморды. Т.е. можно брать за образец что нужно слать в  `~/post` что бы произвести действие

`{"pkg":"post","action":"dev_pwrswitch","data":{"dev_pwrswitch":false}}` - лампа выкл

`{"pkg":"post","action":"dev_brightness","data":{"dev_brightness":11}}` - установить яркость в 11

`{"pkg":"post","action":"eff_sw_next","data":{"eff_sw_next":null}}` - переключить на следующий эффект

`"pkg":"post"` можно из тела отпустить, это специфика вебсокет интерфейса

 ну и пример сложной команды настройки параметров лед-матрицы - шлем в  `~/post`
```
{"action":"display_ws2812","data":{"dtype":0,"mx_gpio":25,"CLmt":2500,"width":16,"height":16,"snake":true,"vflip":false,"vertical":false,"hflip":false,"wcnt":1,"hcnt":2,"tsnake":false,"tvflip":false,"tvertical":false,"thflip":false}}
```
 в ответ в топик `~/pub/value` вернется
```
[{"dtype":0,"mx_gpio":25,"CLmt":2500,"width":16,"height":16,"snake":true,"vflip":false,"vertical":false,"hflip":false,"wcnt":1,"hcnt":2,"tsnake":false,"tvflip":false,"tvertical":false,"thflip":false}]
```

vortigont, 2024-01-05:
можно тоже самое слать в отдельные топики, которые по сути это акшен с замененными разделителями
`~/pub/set/dev/pwrswitch`
{"data":{"dev_pwrswitch":false}} - лампа выкл

`~/pub/set/dev/brightness`
{"data":{"dev_brightness":11}} - установить яркость в 11

`~/pub/set/eff/sw/next`
{} - переключить на следующий эффект, тело можно оставить пустым

`~/pub/get/dev/brightness`
{} - запросить текущую яркость, ответ придет в топик `~/pub/value` в формате как выше в сообщении

BashSer, 2024-01-05:
> > концепт заключается в том, что есть одна точка входа и дальше формируется один джейсон для этого входа
> 
> без проблем, эта точка `~/post`, туда и отправляете джейсоны в теле которых указан акшен и параметры к нему.

> я опять вас не пойму, то вы говорите вам одну точку входа, то теперь разные модули. Но в любом случае можно слать и так и так, результат будет один, это сделано для удобства интеграций, т.к. я в [...]

Тогда получается невалидный джейсон с двумя экшенами, как ранее и было написано. Потому что указывается один топик, который получает команды. Нельзя указать разные топики на каждое действие. Грубо говоря, один топик = один модуль управления. У друга подключили более 15-20 разных устройств купленные в разное время(на разных чипах) в пределах одной квартиры и нигде нет такой схемы - я просто не знаю как донести мысль =(
Повторюсь - как корректно отправить значение на ~/set/ одним джейсоном с вкл\выкл+номер эффекта? Или в post с несколькими экшенами? Не в несколько разных топиков, а в один одной командой?
Вот примеры:
Есть устройство в виде трёх клавишного выключателя - получается у него три точки входа для команд ```~/switch/left/set```, ```~/switch/center/set``` и ```~/switch/right/set``` под каждую кнопку, т.е. три управлялки в одном устройстве, три топика на который надо слать команды в зависимости какая кнопка нужна, что в итоге даёт три самостоятельные сущности
Если у выключателя\реле помимо свитчей есть температура\влажность, то они будут в отдельных сущностях ```~/switch/temp``` и ```~/switch/hum```, которые можно отдельно опрашивать и они получаются самостоятельными температурой и влажностью.
Далее есть "общие" топики в которых идёт обмен кто какая группа, всякие метрики и т.д.
В данном случае получается, что у нас есть множество устройств - выключатель самостоятельная сущность, яркость ещё одна сущность, эффект другая сущность, переключить эффект десятая сущность и т.д. Получается не одно устройство с несколькими модулями, а сотня самостоятельных устройств со своими топиками под которые надо плодить множества сущностей и потом их отдельно соединять, т.е. костылять под них. Попробую в виде дерева изобразить как я это понимаю:
<details>

[image]

</details>

> нигде, нет его. Это рудимент от старого кода, нужно написать новый. Если интересно, то смотрите [main.cpp](https://github.com/vortigont/FireLamp_JeeUI/blob/master/src/main.cpp), ищите функцию [...]

Благодарю, вот это и хотелось посмотреть =)

> это неверно, в валуе публикуются все **значения** каких либо контролов в вебюи, напр изменение яркости, выбор эффекта и пр. У вас [explorer](http://mqtt-explorer.com/) есть? Я там всё смотрю. [...]

При прослушке ```~/pub/value``` все эти сообщения начинают появляться только после того как в веб-интерфейсе один раз зайти во вкладку MQTT. Если один раз после перезагрузки не зайти в эту вкладку, то будет тишина в ```~/pub/value``. Как будто бы нужно просто ОДИН раз зайти во вкладку веб-интерфейса, что бы очнулся MQTT - только после этого в топике все вышеперечисленные сообщения начинают публиковаться ровно в том виде как Вы и продемонстрировали. Спасибо за наводку про mqtt-explorer - не знал =)

<details>
<summary>оффтоп</summary>

Что-то заинтересовало всё по новой, достал запылившийся шлюз, поставил все дополнения и интеграции, начал из кладовки доставать разные zigbee модули и подключать их. Всем HA поставил, а сам у себя всё сломал и никак настроения не было заняться вновь. Разбудили во мне какую-то мотивацию и выздоравливайте там! =)

</details>

BashSer, 2024-01-05:
Случайно по другому делу наткнулся на [это](https://github.com/MishanyaTS/FieryLedLamp_New_WEB/blob/main/FieryLedLamp_New_WEB/FieryLedLamp_New_WEB/MqttManager.h)
Похоже это связанно с новогодним [саботажем](https://github.com/MishanyaTS/FieryLedLamp_New_WEB/blob/main/FieryLedLamp_New_WEB/FieryLedLamp_New_WEB/MqttManager.h) - не знаю точно и не буду разбираться, мы не о политике, а о коде и на форуме есть ссылка на [гугл.доки](https://docs.google.com/document/d/1YkmxJlh-6DlPpsllkVFMBwpEFf3RNqiA/edit?usp=sharing&ouid=115116093970636946286&rtpof=true&sd=true) где по MQTT дружат лампу с Яндекс.Алисой. Есть подозрение, что в первой ссылке адекватно описана работа по MQTT, раз лампа умеет дружить с Алисой. Даже если это не принесёт пользы, то лучше я об этом упомяну чем не упомяну :)

vortigont, 2024-01-06:
> Повторюсь - как корректно отправить значение на ~/set/ одним джейсоном с вкл\выкл+номер эффекта? Или в post с несколькими экшенами? Не в несколько разных топиков, а в один одной командой?

никак. Такой функционал не закладывался ибо как по мне то это несколько странно. Сущность "эффект" никак не зависит от сущности "включение лампы", либо "яркость матрицы", либо "микрофон" и пр. Это как включить телевизор с пульта одной кнопкой сразу на нужный канал, с определенной громкостью и заданной яркостью. Вам придется иметь либо тысячу кнопок либо делать таки макрофункцию когда одна кномка генерирует 3 разных команды - включение, переключение канала, выставление громкости.

Или беря ваш же пример одной командой включить все 3 выключателя отправив один запрос в один топик.
У вас также есть множество отдельных сущностей "center", "left", "right". Каждой сущностью вы управляете отдельно, включаете, выключаете, снимаете статус. Вы же не пытаетесь одной командой включить все 3 выключателя?

Также  и здесь вам нужно отправить отдельные команды на включение лампы, на переключение эффекта, на установку яркости, подергать отдельно набором контролов для эффекта (а набор и тип этих контролов может быть разный у каждого эффекта). Это можно делать как через один общий топик `~/post` так **И** через отдельные топики `~set/#`. **И** то **И** то будет работать - выбирайте вариант который вам удобнее.

Почему это нелья смешать в одну акшен-команду - приведу вам простой пример, напр у эффекта есть 3 контрола,  скорость, масштаб, и какая-нибудь шкала. Значение всех контролов определяется ключем "value", как уместить в одну команду 3 разных ключа "value" я смутно представляю.

Я может не очень понимаю чего вы пытаесь получить в итоге, но зачем включать лампу и выставлять номер эффекта именно в **одной** команде? Нельзя ВЕСЬ функционал лампы уместить в одну команду. Я уже говорил, различных ключей в апи очень много, многие их них повторяются и смысл их меняется в зависимости от акшена. Например установка параметров матрицы из ленты и хаб75 панели использует одинаковые параметры "ширина", "высота", внешняя периферия использует парметры настройки gpio, которые могу перекрываться. Какой смысл тогда в разделении по топикам и сущностям если вы хотите все делать в одной команде?

> Тогда получается невалидный джейсон с двумя экшенами, как ранее и было написано. Потому что указывается один топик, который получает команды.

Не понимаю стремления делать все именно в одной команде, в чем сложность послать несколько команд в один топик `~/post`? Поясните, может я как-то не так себе это вижу?

Впринципе я могу подумать и реализовать например такой функционал:
на топик `~/post` можно будет отправить джейсон, который состоит из массива объектов, каждый из которых это акшен-команда со своим набором параметров и они будут выполненны по очереди.
Например:
```
 [
 {"action":"dev_pwrswitch","data":{"dev_pwrswitch":true}},
 {"action":"eff_sw_idx","data":{"eff_sw_idx":10}},
 {"action":"dev_brightness","data":{"dev_brightness":11}}
]
```
такое сообщение включит лампу, переключит эффект на 10, выставит яркость в 11 одной командой в один топик, такой вариант вам будет более удобен?

> При прослушке ~/pub/value все эти сообщения начинают появляться только после того как в веб-интерфейсе один раз зайти во вкладку MQTT

Блин, это технически невозможно :) Вы когда туда заходите то просто заходите или нажимаете кнопку 'save'?
Проведите простой эксперимент - закройте страницу брайзера с вебмордой, выдерните питание контроллера, запустите эксплорер, подключитесь к своему мкутт, включите контроллер. Брайзер открывать не надо, у вас должна придти публикация в системный топик где будет параметр uptime со значением в несколько секунд. Это значит что контроллер подключился к мкутт серверу и будет слать туда сообщения. Если этого не произойдет, значит контроллер по какой-то причине вообще не подключился к серверу. Подробности надо искать в логах.

> где по MQTT дружат лампу с Яндекс.Алисой
никогда не пользовался "алисой", но, думаю, если есть АПИ, то подружить можно что угодно.
Я посмотрел в файл по ссылке, там используется простая система отправки одиночных текстовых команд в один топик. Т.е. включить лампу `P_ON`, переключить эффект `EFF0` и т.п.

BashSer, 2024-01-06:
> Сущность "эффект" никак не зависит от сущности "включение лампы", либо "яркость матрицы", либо "микрофон" и пр.
> Вам придется иметь либо тысячу кнопок либо делать таки макрофункцию когда одна кномка генерирует 3 разных команды - включение, переключение канала, выставление громкости.

В этом я заключаются костыли сейчас, из-за множества разных топиков. Можно расписать для HA файлик автоматизации с тысячей кнопок где одна кнопка = публикация отдельного сообщения в отдельный топик, но его будет неудобно поддерживать и редактировать. Такой костыльный файлик никак не войдёт в мастер HA

> Вы же не пытаетесь одной командой включить все 3 выключателя?

Примерно так и есть =) Zigbee2MQTT [web](https://www.zigbee2mqtt.io/supported-devices/) и [git](https://github.com/Koenkk/zigbee2mqtt) ("Currently 3386 devices are supported from 400 different vendors"). В течении дня через issue добавляется поддержка нового неизвестного устройства в дев-ветку и потом мигрирует в мастер после проверок и правок

<details>

[image]

[image]

</details>

> подергать отдельно набором контролов для эффекта (а набор и тип этих контролов может быть разный у каждого эффекта) 
> Почему это нелья смешать в одну акшен-команду - ... Значение всех контролов определяется ключем "value", как уместить в одну команду 3 разных ключа "value" я смутно представляю

Объект эффект в котором генерируется список доступных контролов, каждого из эффектов. Например, эффект n имеет контролы cntrl_1: 0-255, cntrl_2: 0-100, etc. Если что-то есть, то добавляем в основное сообщение, иначе упускаем атрибут([HA Light MQTT](https://www.home-assistant.io/integrations/light.mqtt/#json-payload)). Развиваем дальше cntrl_1, cntrl_2 заменяем на более понятные scale, speed и делаем их универсальными для других эффектов, а эффект будет иметь вид effect: 1, name: "wow_effect_3000", speed: 0-100, scale: 0-100. Когда будем парсить атрибуты, то станет понятно какой эффект, какие ползунки доступны, как формировать сообщение(if defined, then...) и что отображать в карточке. Профит - если добавляется\изменяется эффект, то он автоматически подсосётся, не надо ковырять файл автоматизаций с тысячей уникальных кнопок =)

> Я может не очень понимаю чего вы пытаесь получить в итоге, но зачем включать лампу и выставлять номер эффекта именно в **одной** команде?
> Не понимаю стремления делать все именно в одной команде, в чем сложность послать несколько последовательных команд в один топик ~/post? Поясните, может я как-то не так себе это вижу?

Стандартные карточки MQTT Light из коробки так работают. При переключении эффекта применяется, тот самый стандартный шаблон([HA Light MQTT](https://www.home-assistant.io/integrations/light.mqtt/#json-payload)). Нельзя в одном объекте прописать разные топики для команд, один объект = один топик, отдельные топики = отдельные команды, макросы отдельных команд = костыли с сотнями объектов призывающие сотни отдельных маленьких команд с прописыванием взаимосвязей, т.е. не подгонка лампы под тысячи устройств и всех домов, а подгонка всех домов под лампу получается. Можно пойти другим путём - скрафтить свою интеграцию и потом добавить в мастер HA, что бы было доступно для всех и при дискавери HA предлагал автоматически добавить и настроить интеграцию заточенную именно под эту лампу, но это более времязатратный способ, зато более подходящий для распространения в люди

<details>

[image]

</details>

> Нельзя ВЕСЬ функционал лампы уместить в одну команду

Объект LED ```{state: on\off,, eff_number: 42, brightnes: 0-100, ifdef scale: 0-100, ifdef speed: 0-100...}``` - включаем\выключаем, дёргаем ползунки
Объект TempHum ```{state: on\off, temp: 20, hum: 50}``` - представляем отдельный объект как подключаемый модуль если он есть для просмотра температуры\влажности
Объект Mic ```{state: on\off, ifdef command: calibrate\conf\amp_gain, ifdef value_for_action: 50....}``` - ещё один подключаемый объект со своими командами, который отсылает свой экшн с ползунком, если он доступен

> Например установка параметров матрицы из ленты и хаб75 панели использует одинаковые параметры "ширина", "высота", внешняя периферия использует парметры настройки gpio, которые могу перекрываться.

Есть параметры, которые могут сохраняться и их не надо отправлять общей командой как раз для "редких" настроек, а настроенные значения читаются из того же дискавери. Однако, можно переписать эти параметры соответствующей  командой в нужном топике. Честно говоря, я бы весь этот fine-tuning и редкие настройки оставил на стороне веб-интерфейса лампы. HA сейчас стал мега доступным и теперь он про удобство, а не про гик переписывание всего кода как 5-10 лет назад или более 10-15 лет лет в openHAB. Т.е. включил лампу, сделал первую настройку если требуется, подключил к сети, авто-дискавери понял кто это в сети появился, сгенерировал объект и везде его прописал или предложил требуемую интеграцию, которая этим займётся. В общем, как это и работает со всеми устройствами сейчас, а если нужна тонкая настройка, то уже используешь спец.ПО, веб-морду и т.д. :) Смысл в том, что есть основное\повседневное использование и подвязка к автоматизациям(будильник, триггеры, выходные), а есть углублённая настройка со своими профессиональными средствами, которые не относятся к повседневной жизни.
Включил диммер, подключил к домашней сети, связал с рабочими\выходными\закат\рассвет\пришёл\ушёл - работает. Хочу влезть во внутренности и что-то переписать? Начинаю перепаивать, использовать соответствующий софт, при [... cut ...]
Был ещё случай, когда просто закрылся Insteon и все устройства умерли, потому что были завязаны на их облаке. Коммунити [HA](https://www.home-assistant.io/blog/2022/04/19/for-insteon-users/) быстро отреагировала и как-то спасли своих погорельцев. Юзвери получили новый опыт, коммунити снова получила большой буст в разработчиках и обычных пользователях, что только даёт развитие HA семи мильными шагами. Активно форкают компоненты и добавляют свои интеграции в мастер, включая не только обычных пользователей, но и большие бренды, которые также подстраиваются под HA

</details>

vortigont, 2024-01-06:
> В этом я заключаются костыли сейчас, из-за множества разных топиков.

ёклмн... еще раз повторяю ОТДЕЛЬНЫМ сообщением - есть **ОДИН** общий топик для **ВСЕХ** команд `~/post`

Остальное вдумчиво почитаю, отвечу позже

vortigont, 2024-01-06:
давайте по одному сообщению разбирать, иначе бардак получается.

> Объект LED `{state: on\off,, eff_number: 42, brightnes: 0-100, ifdef scale: 0-100, ifdef speed: 0-100...}` - включаем\выключаем, дёргаем ползунки

какое отношение вкл/выкл имеет к эффекту и яркости? Пылесос можно сделать вкл/выкл, какую ему задать яркость и эффект? Обычную лампочку можно сделать вкл/выкл, задать яркость, какой ей можно задать эффект?
`ifdef scale: 0-100` это всё хорошо, но если у эффекта нет scale а есть, напр `number of stars`, вы же не будете править шаблон и добавлять еще один if?

> Объект Mic `{state: on\off, ifdef command: calibrate\conf\amp_gain, ifdef value_for_action: 50....}` - ещё один подключаемый объект со своими командами, который отсылает свой экшн с ползунком, если он доступен

что в вашем понятии "объект" и что "команда"?
В моем понятии объект это "телевизор", он умеет принимать МНОГО команд - переключение каналов, звук, яркость, вызвать меню, установить таймер и т.п. Принимать одну команду в которой описан ПОЛНЫЙ статус объекта целиком он не умеет.

vortigont, 2024-01-06:
немного "теории" о том как написан мой проект и как в нем устроена вебморда и мкутт.
Вебморда использует вебсокет и гоняет джейсон объекты между контроллером и браузером. В объектах может быть структурированная вебстраница с элементами контролов - переключатели, поля, ползунки и т.п., могут быть команды - чекрыжек такой-то изменил статус, могут быть значения - чекрыжек такой-то имеет значение такое-то. Могут быть сложные объекты типа состояние группы чекрыжиков и полей в едином блоке. Абсолютно теже самые объекты бегают и через мкутт - команды и состояния. Это позволяет избежать дублирования кода и функционала. В вебморде можно поменять параметр, а в мкутт - нет. А добавьте в мкутт еще одну штуку которая есть в вебморде и т.п. Описаный однажды объект, его свойства и поведение становятся доступны через 3 разных апи - вебсоке, мкутт, хттп рест (с ограничениями).

В чем смысл отдельных команд - они управляют состоянием одиночных объектов. Рычажек выключатель лампы, ползунок управления яркостью, ползунок какого-либо из контролов у эффекта. Вы дернули обхект - отработала ОДНА команда поведения конкретно этого объекта. Отрегулировать яркость можно независимот от того какой сейчас установлен эффект и наоборот - переключая эффект, вам наплевать какая сейчас у лампы яркость.

Почему я невзлюбил ХА и мкутт так это как раз потому что те кто занимаются интеграциями с "умными домами" всегда пытаются перекроить устройство под свою модель управления, а не подстроить управление под модель работы устройства. Лампы должны быть некими "стандартными" иначе работать не будет. Но с другой стороны вся разработка ХА состоит из постоянного добавления десятков новых устройств. Так зачем добавлять "поддержку нового устройства" если оно будет использовать стандартные ХАшные шаблоны? Должно всё работать само и сразу, никаких новых устройст добавлять в ХА не надо :))) Шучу конечно.

Теперь смотрим дальше.
Вот вы захотели поменять яркость на лампе? Какую команду вы хотите послать?
`{state: on\off,, eff_number: 42, brightnes: 0-100, ifdef scale: 0-100, ifdef speed: 0-100...}` - ну пускай эту.
Т.е. что бы просто поменять яркость вам нужно в команду включить текущее состояние питания лампы (ну понятно что on), текущий её эффект, т.к. вам нужно просто яркость поменять, а эффект оставить прежним и плюс полный набор контролов. Маразм? Ну ладно, допустим ХА все эти состояния у себя отслеживает и таки действительно может эти данные впихнуть в тело одной команды. Ниже распишу почему это абсурд для управления сложным объектом типа лампа/гирлянда/информер/телевизор.

vortigont, 2024-01-06:
> Тут больше вопрос не об удобности и подгонке костылей под лампу, а про универсальность и соответствие требованиям другим домам и устройствам.

я этих требований пока не увидел. По [ссылке](https://www.home-assistant.io/integrations/light.mqtt/#json-payload) описан **простейший** джейсон объект с 4мя параметрами и ямл-файл что бы его "понять".
Нет информации что это единственно возможный шаблон, что набор параметров должен быть именно таким и это всё должно передаваться в едином объекте одним сообщением. Это тупо какой-то простейший экзампл ямл-файла.

Так я еще раз повторю свой вопрос - существует ли в ХА техническая возможность отправлять **отдельные** команды для управления устройством? Не всё-в-одном-за-раз, а отдельно включать, отдельно менять яркость, отдельно переключать эффекты, отдельно передвать сложный объект с расписанием будильника по дням недели?

А теперь почему это всё неприменимо для управления таким сложным устройвом как "магическая лампа", а не тупая ргб-лампочка от филипс. 

Допустим мы согласились с тем что нужно работать с командой все-в-одном вида
`{"state": "on", "brightness": 255, "color": [255, 255, 255], "effect": "rainbow"}`

У лампы есть дофигища внутренних объектов, которые взаимодействуют между собой и реагируют на одни и те же события, иногда одинаково, иногда по-разному. Синхронизированная работа компонентов это адов труд в проекте.
К примеру есть экранчик ТМ1637, который показывает время, при вкл/выкл выводит текст On/Off, при смене яркости выводит значение яркости, при смене эффекта выводит номер эффекта. Есть еще "бегущая строка" которая делает тоже самое  на самой матрице (у меня сейчас она не работает, но не суть).
И тут лампа получает такое сообщение - `{"state": "on", "brightness": 255, "color": [255, 255, 255], "effect": "rainbow"}`, где хотели выставить только яркость.
Что должно отобразиться на дисплее? Статус On? Яркость или номер эффекта и все его контролы? Или все по очереди? Абсурд? Да :)
Идем дальше - есть МП3 плеер, который ловит события типа "смена эффекта" и начинает играть музыку. Что делать ему в этом случае? Прерывать дорожку и начинать играть её же заново? И т.д. и т.п.

Оно всё выглядит легко и красиво когда управляешь "выключателем" в котором тупо 3 кнопки и 2 состояния. Тогда удобно всё 3 состояния класть в одну команду и не париться. Но если вы хотите управлять сложным устройством то без множества команд, топиков и какой-то внятной структуры не обойтись.

Я готов всё это обсуждать, но нужно что бы был понятен масштаб задачи :)

BashSer, 2024-01-06:
> ifdef scale: 0-100 это всё хорошо, но если у эффекта нет scale а есть, напр number of stars, вы же не будете править шаблон и добавлять еще один if?
Я вам про универсальность, а вы снова про уникальность =)
Зачем уникальный контрол, если можно подбить все контролы под пять ползунков и дальше либо есть контрол, либо его нет

> Так зачем добавлять "поддержку нового устройства" если оно будет использовать стандартные ХАшные шаблоны? Должно всё работать само и сразу, никаких новых устройст добавлять в ХА не надо :))) Шучу конечно.

Вот так выглядит коммит "поддержки нового устройства" от очередного азиата\ноунейм бренда\пользователя\т.д., потому что да - они все стандартные и работают сразу, прямо как вы и сказали =) Шучу конечно.
[image]

> Так я еще раз повторю свой вопрос - существует ли в ХА техническая возможность отправлять отдельные команды для управления устройством? Не всё-в-одном-за-раз, а отдельно включать, отдельно менять [...]

Технически - да, можно. Удобно и будет ли жить в перспективе - не уверен
Я примерно с этого и начал, что стандартная карточка MQTT Light так не работает. Есть один топик для отправки команд и сообщение генерируется приводя в примеры документацию и другие рабочие устройства где везде только один топик для команд. Делать что-то отдельно - это соответственно отдельная кнопка. Описывать отдельно каждую кнопку не универсально - захламляет, неудобно, поддержки не будет. Если нужна отдельная кнопка под каждую команду, то проще как-то свою интеграцию писать, что бы хоть как-то отслеживать изменения очередной кнопки

> Что должно отобразиться на дисплее? Статус On? Яркость или номер эффекта и все его контролы? Или все по очереди? Абсурд? Да :)

Так это же другой топик как подключаемый модуль и соответственно объект. Выше я это тоже упоминал. Таким топиком может быть и датчик температуры\влажности со своим набором, как выше был пример

>Идем дальше - есть МП3 плеер, который ловит события типа "смена эффекта" и начинает играть музыку. Что делать ему в этом случае? Прерывать дорожку и начинать играть её же заново? И т.д. и т.п.

Зачем нам опять же лезть в другой объект? Будет объект мп3-плеер со своим набором кнопок для включения музыки или что он там делает, примеры выше были опять же

> Оно всё выглядит легко и красиво когда управляешь "выключателем" в котором тупо 3 кнопки и 2 состояния. Тогда удобно всё 3 состояния класть в одну команду и не париться. Но если вы хотите управлять сложным устройством то без множества команд, топиков и какой-то внятной структуры не обойтись.

Так легко и должно быть для обычного пользователя =)
Зачем усложнять? Зачем такая тонкая настройка и главный вопрос для кого? Я могу понять, если это было бы "сложное устройство" в котором надо каждый час что-то менять и тогда была бы причина всё подгонять, но большая часть делается единожды, а весь бытовой функционал и заключается в простоте. Что бы свет "следовал" за тобой, дом приветствовал тебя, прощался и будит тебя утром, а к ночи приглушал свет намекая что пора спать, сам открывался и закрывался при входе\уходе, убирался тоже сам, только напоминая об обслуживании и вот это всё, а не заходить в какую-то настройку третей луны каждые пять минут и менять её =)

<details>

[image]

При этом можно тут же перейти в режим кода и более подробно переписать, если требуется

</details>

> Я готов всё это обсуждать, но нужно что бы был понятен масштаб задачи :)

Я только за, но кажется у нас разные восприятия данного устройства =(
Вашу точку зрения я пока что понимаю так - "это уникальный продукт, с уникальными настройками, где всё можно отдельно крутить, пусть другие под меня подстраиваются, а не я под общепринятое"
Я пытаюсь донести - "было бы круто сделать универсальным и общепринятым, чтобы запуллить в мастер и даже спустя годы любой пользователь сможет прошить esp и не танцевать с бубном"

<details>
<summary>wled</summary>

Сейчас внезапно начал подробнее читать про WLED и прошил esp8266 стандартной прошивкой подключив ту же самую матрицу на "соплях". Есть странички про [json](https://kno.wled.ge/interfaces/json-api/), [mqtt](https://kno.wled.ge/interfaces/mqtt/), websocket, http и т.д. (слева в меню)
В [списке эффектов](https://kno.wled.ge/features/effects/) также почти одинаковые контролы, а в веб-интерфейсе и вовсе три универсальных ползунка, которые либо есть, либо нет
В веб-морде уже есть подключаемые модули в виде кнопок, ИК, реле, микрофона, температуры, датчики движения, маленькие oled дисплеи с энкодером(сегментного дисплея, SD-карты сходу не нашёл), но возможно стоит упомянуть об этом и со временем добавят недостающее
[image]

Оттуда же настраивается кол-во и тип светодиодов, лент, генератор 2D матриц, групп и бла-бла-бла
Используется стандартная карточка света
[image]

Самостоятельно о себе рассказывает и делится на объекты. Вангую, что после подключения кнопки\реле\другой модуль - точно так же сгенерируется объект со своим топиком для управления, как я ранее описывал
[image]

[image]

[image]

Про wled упоминал ранее, потому что вроде бы библиотека какая-то использовалась с подобным названием, но arduino и esp пока слишком далеки от меня - могу ошибаться и быть не правым

Может оттуда можно какие-то полезности подёргать, что бы не изобретать колесо по-новой? =)

</details>

vortigont, 2024-01-06:
создал группу в телеге t.me/lampdevs

vortigont, 2024-01-06:
> Я вам про универсальность, а вы снова про уникальность =)
> Зачем уникальный контрол, если можно подбить все контролы под пять ползунков

вы по эффектам то в лампе хоть прошлись, полистали? Видели разнообразие и количество контролов там?
Можно конечно кастрировать всё до стандартных 5ти контролов, но это уже деградация получается в угоду простоте управления. WLED выбрал эту стартегию, поэтому эффекты у них самые убогие и простые. Делать еще один WLED смысла нет, лучше сразу взять WLED.

> Вот так выглядит коммит "поддержки нового устройства" от очередного азиата

приведите, пожалуйста нормальную ссылку на комит а не скриншот. Толку то мне от скриншота.

> все стандартные и работают сразу, прямо как вы и сказали =) Шучу конечно.

так поясните мне смысл отдельного нового комита от очередного азиата если там стандартная карточка. Я серьезно спрашиваю - что нужно комитить в ХА для того что бы использовать **его же** стандартные карточки?

vortigont, 2024-01-06:
> Есть один топик для отправки команд и сообщение генерируется приводя в примеры документацию и другие рабочие устройства где везде только один топик для команд.

я терпеливый, я вам 4й раз приведу выделенно тот же тезис который вы почему-то с завидной регулярностью игнорируете

есть **ОДИН** общий топик для **ВСЕХ** команд `~/post`

ответте, пожалуйста, конкретно на это сообжение - это точто вам нужно или нет?

vortigont, 2024-01-06:
иду на страничку [WLED MQTT](https://kno.wled.ge/interfaces/mqtt/)

и вижу там что влед слушает 6 разных топиков
**отдельный** топик на вкл/выкл,яркость **[mqttDeviceTopic]**

**отдельный** топик на цвет **[mqttDeviceTopic]/col**

**отдельный** топик на джейсон апи **[mqttDeviceTopic]/api**

в который можно плеваться джейсонами в соответсвие с его апи. Т.е. никакой единой точки входа нет, есть вход для апи, собссно как у меня и реализованно. Упоминается что мкутт они сейчас переделывают ради ХА опять же, но тема куда ведет ссылка открыта в 2019 году, последнее сообщение там от марта прошлого года.
Я еще полазию там, погляжу где у них шаблоны для ХА в репе, но пока чуда я не нашел )

BashSer, 2024-01-06:
> я терпеливый, я вам 4й раз приведу выделенно тот же тезис который вы почему-то с завидной регулярностью игнорируете
> есть **ОДИН** общий топик для **ВСЕХ** команд `~/post`
> ответте, пожалуйста, конкретно на это сообщение - это точто вам нужно или нет?

Да, это тот топик, но он мне не нужен, потому что его структура позволяет принимать только один экшн, а не набор базовых параметров. Понял с первого раза и использовал его ещё до того как Вы на него указали.
Вы точно также настойчиво игнорируете, что в синтаксисе есть только ОДНА переменная [command_topic](https://www.home-assistant.io/integrations/light.mqtt/#command_topic). Ну вот нельзя сделать в одном объекте лампы 50 топиков. Это будет 50 отдельных объектов с узкоспециализированной кнопкой, что в свою очередь нецелесообразно. Не нужно это в быту, это не программатор. Это лампа. Я могу быть не прав, нужно использовать другой шаблон для "сложного устройства", "программатора", "станка" или ещё чего-нибудь, но я вижу это как эстетически красивую и хорошую лампу. Сценарии использования - будильник, восход, включись, выключись, уведомления, яркость, переключить эффект с привязкой к другим событиям - красиво, удобно, приятно, входит в часть "быта" и воспринимается как нечто обычное.
Прощу прощения, не вижу смысл продолжать беседу в подобном ключе.

>и вижу там что влед слушает 6 разных топиков

Да, да, и да. Я же это всё говорил в самом начале и предложил примеры на основе рабочих устройств как можно разделить эти топики. У вас получается по каждой точке на каждую кнопку, а не основные топики команды\конфиг\статус. На любой странице документации справа есть кнопка посмотреть на гитхабе - вот прямая ссылка на [mqtt light](https://github.com/home-assistant/core/tree/dev/homeassistant/components/mqtt/light). Уровнем выше все остальные репозитории куда можно сделать пулл реквест, что бы Ваш продукт пошёл в массы. Вряд ли чудо найдёте, обычный труд многих пользователей за длительное время, но в этом Вы уже лучше чем я разбираетесь.

Ещё раз приношу свои искренние извинения за Ваше потраченное время. Мы не слышим друг друга и преследуем разные цели. Вы стараетесь над узкоспециализированным и сложным устройством, а я про доступность и удобства, чтобы это стало "обычным". ИМХО будущее, прогресс и технологии определяются не тем, что уникальный индивид из большинства может гвоздём спаять чудо, а общий уровень большинства позволяет простым паяльником делать средние достижения. Ваша лампа для гиков(в хорошем смысле), а не для обычного пользователя, увы. Вы хороший человек и успехов Вам! =)

closed 2024-01-06 by BashSer

vortigont, 2024-01-06:
Что-то вы быстро сдались, мы даже о терминах еще не договорились... ну как знаете :(

vortigont, 2024-01-06:
кстати, вот как реализованна поддержка WLED в HA - [components
/wled](https://github.com/home-assistant/core/tree/master/homeassistant/components/wled)

т.е. там пачка питоновских скриптов, которые формируют джейсон-объекты для работы с АПИ ВЛЕД и шлют их в один апи-топик в мкутт. В частности вот [здесь](https://github.com/home-assistant/core/blob/master/homeassistant/components/wled/light.py) как раз управление светом.
Нет там никаких "стандартных" [MQTT Light ](https://www.home-assistant.io/integrations/light.mqtt/) от ХА, а есть жесткий кастом который работает с АПИ :)