vortigont/FireLamp_JeeUI #48: Эффект "250, Часы" - некорректное отображение

BashSer, 2024-01-04:
Добрый вечер !

При включении эффекта анимация странного падающего тетриса, после чего время 17:24 отображается следующим образом:
[image]

Предполагаю, что виноваты настройки в веб-интерфейсе Settings -> Настройка LED-дисплея -> включены "змейка", "отзеркалить столбцы", "вертикальная лента". Не знаю по какой причине, но три матрицы, которые у меня были все с повёрнутым набок изображением и в никакой прошивке это не исправлялось плюс всегда были перепутаны RGB каналы, но это уже чинилось в коде. Пробовал на esp8266 и esp32, пробовал в соответствующем куске кода менять значения с направлением матрицы и только настройка через веб-интерфейс помогла повернуть изображение
[image]

vortigont, 2024-01-04:
Привет. Этот эффект расчитан на широкие матрицы, поставьте контрол "масштаб" в единицу, тогда влезут 2 цифри всего.
Настройки дисплея влияют на все эффекты одинаково, настройте один правильно, остальные тоже заработают нормально.
Какую версию вы пробовали на 8266? 8266 уже давно не поддерживается.

BashSer, 2024-01-04:
Сейчас всё на esp32, релизная версия 3.4.0, которую я теперь не могу найти у Вас =D Но она осталась на ПК
Поигрался с настройками, масштаб ставил разные, результат примерно такой же, время 18:47 или около того:

https://github.com/vortigont/FireLamp_JeeUI/assets/37932617/141701e8-ef14-452f-b664-a182e0ffd370

vortigont, 2024-01-04:
если вы про [этот](https://github.com/vortigont/FireLamp_JeeUI/releases/tag/v3.4.0) релиз, то там часы отображаются вверх ногами, что не отменяет того факта что на матрицу 16х16 они не влезут ни при каких раскладах. Минимум нужна панель 32х16.
Можете попробовать текущий мастер, но там могут быть некоторые нестабильности.

BashSer, 2024-01-04:
Будем ждать решения #25  - добавление мелкого шрифта и отображение поверх эффекта звучит хорошо.
Стыковать много матриц нет в планах(я даже не представляю где взять вменяемый БП для подобного), только "лампа" и что-то с гирляндами, а в гирляндах те же самые 16 светодиодов можно расположить на расстоянии 20см, а это уже 3 метра с лишним - для отображения информации не подойдёт, но эффекты на окнах\доме будет выглядит неплохо =)
Благодарю!

closed 2024-01-04 by BashSer