vortigont/FireLamp_JeeUI #105: update-DEV-from-Git.cmd

andy040670, 2024-06-15:
А куда делся update-DEV-from-Git.cmd ???
В билдере теперь не возможно переключиться между ветками и обновить локальный репо.
Закидываю файл с прошлого среза, получаю:
`Your choice [Ваш выбор]: ► 1
The system cannot find the path specified.
The system cannot find the path specified.
'witch' is not recognized as an internal or external command,
operable program or batch file.
'м' is not recognized as an internal or external command,
operable program or batch file.
'l' is not recognized as an internal or external command,
operable program or batch file.
'�' is not recognized as an internal or external command,
operable program or batch file.
HEAD is now at a04e984 Merge pull request #104 from vortigont/tg
Already up to date.`

andy040670, 2024-06-15:
Ссылка на форум в вебморде была не лишней.
Мне вот телега нафиг не нужна, я ей не пользуюсь, хоть и бот мой там трудится, собирает нужную информацию с моего сервера, но без необходимости туда даже не заглядываю.

vortigont, 2024-06-17:
@andy040670 ветка `DEV` уже давно не используется и никаких изменений туда не вливается. Отдельные новые функции я вливаю из разных веток через пулл-реквесты. Переключать ветки из скрипта больше не получится, т.к. заранее имя ветки неизвестно.
Сам cmd файл я могу вернуть если без него билдер не работает, но его функционал по сути сводится только к `git pull`.

Ссылку на форум поменял, т.к. на форуме давно уже ничего не происходит толком.
Если пользовались, оставлю обе.

andy040670, 2024-06-17:
> Сам cmd файл я могу вернуть если без него билдер не работает, но его функционал по сути сводится только к `git pull`.

Там ещё архиполезная функция "сброс локального репозитория"
`	%gitcmd% reset --hard
	rem "%ProgramFiles%\Git\bin\git" checkout .
	%gitcmd% pull`
В самом билдере первый пункт переименовать в "Обновить локальный репозиторий (Мастер ветка)"
Второй пункт убрать из за неактуальности.

closed 2024-06-21 by vortigont

andy040670, 2024-06-21:
Ссылки `<span class="menu-title"><a href="https://community.alexgyver.ru/threads/wifi-lampa-girljanda-informer-ws2812-hub75-obsuzhdenie-proshivki-firelamp_embui.7257/" target="_blank" rel="noopener noreferrer">Gyver</a></span>` в data/index.html.gz нет, соответственно в UI тоже её нет.

vortigont, 2024-06-21:
html я не перепаковывал. Обновится вместе джейсонами при следующем изменении элементов в вебюи.
Можете пока положить хтмлку на ФС как есть и удалить имеющийся там .gz

andy040670, 2024-06-21:
А, понятно. 
Да, я ручками всё сделал (перепаковал .gz).

commit 27f7a1cf (2024-06-21): вернул батник update-DEV-from-Git.cmd

по просьбе из #105

Closes #105
commit 3c9009a1 (2024-06-21): возвращена ссылка на форум гайвера в шаблоне WebUI
Closes #105