vortigont/FireLamp_JeeUI #111: Большие цифры и ночной режим для хаб панелей 64х32 и 128х64

dina16121974, 2024-06-28:
Так как я ,не вижу в библиотеках ширину шрифтов. Прошу Вас ,на ваш вкус выбрать максимально большие шрифты ,которые умешаються ,на панели 64х32 и 128х64. И по возможности ,добавить ночной режим для часов с... и до... Чтобы в это время на экране были только цифры часов.

vortigont, 2024-06-29:
Ну вот же размер в пикселях указан `BBX Width 11, Heigh 16`

[image]

Покопайтесь на страничках
https://github.com/olikraus/u8g2/wiki/fntlist99
https://github.com/olikraus/u8g2/wiki/fntlistall

vortigont, 2024-06-29:
Шрифты в прошивке можно менять в файле `widgets.cpp` в 106й стоке. Порядок совпадает с порядком в выпадающем списке в настройках. Можно поменять, проверить как выглядит.

dina16121974, 2024-06-29:
Спасибо ,сейчас размеры буду глядеть

dina16121974, 2024-06-30:
u8g2_font_spleen_12x24_mn это для 64х32
u8g2_font_mystery_quest_48_tn это для 128х64

dina16121974, 2024-06-30:
Ну и контурные добавить
u8g2_font_maniac_tn
u8g2_font_lucasarts_scumm_subtitle_o_tn

dina16121974, 2024-06-30:
u8g2_font_mystery_quest_48_tn заменить на u8g2_font_mystery_quest_56_tn
и добавить u8g2_font_mystery_quest_32_tn и u8g2_font_osb21_tn
Это я все проверил ,у нас для 128х64 ,вообще куцо. Не ожидал от Вас такой прыти ,думал успею.

commit e07c7561 (2024-07-02): добавлены большие шрифты для часов

Closes #111
closed 2024-07-02 by vortigont