vortigont/FireLamp_JeeUI #172: не собираеться последний релиз

dina16121974, 2025-05-24:
*  Выполнение задачи: C:\Users\DENIS\.platformio\penv\Scripts\platformio.exe run --environment esp32S3 

Processing esp32S3 (board: lolin_s3; framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/2025.05.40/platform-espressif32.zip)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing file://C:\Users\DENIS\.platformio\tools\toolchain-xtensa-esp-elf
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Tool Manager: Installing file://C:\Users\DENIS\.platformio\tools\tool-mklittlefs
Tool Manager: tool-mklittlefs@3.2.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/xtensa-esp-elf-14.2.0_20241119.zip
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/mklittlefs-3.2.0.zip
Unpacking  [####################################]  100%
Tool Manager: tool-mklittlefs@3.2.0 has been installed!
Library Manager: Installing vortigont/EmbUI @ ~4.2.2
UnknownPackageError: Could not find the package with 'vortigont/EmbUI @ ~4.2.2' requirements for your system 'windows_amd64'

 *  Процесс терминала "C:\Users\DENIS\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32S3'" был завершен с кодом выхода 1. 
 *  Терминал будет повторно использоваться задачами. Чтобы закрыть его, нажмите любую клавишу.

dina16121974, 2025-05-24:
Не нравиться 4.2.2

vortigont, 2025-05-24:
я не знаю что это за ошибка, вероятна бага или в платформио или в плагине вскода. У меня тоже _иногда_ выскакивает только говорит что  "for your system 'linux_amd64'". Если из консоли командой собирать, то нормально работает.

vortigont, 2025-05-24:
ясен пень что версии либы под винду быть не может, потому что она под есп32 :)))

dina16121974, 2025-05-24:
Чет я запутался. какой командой?
у меня такие есть
 git --version
  git pull обновление репозитария
  git status
  pio --version
  pio pkg update обновление библиотек
  pio run -e esp32debug -t upload прошивка с логом
  pio device monitor -e esp32debug вывод лога
  pio run -e esp32 -t upload прошивка 
  pio run -e esp32S3 -t upload прошивка 
  pio run -e esp32 -t uploadfs прошивка Фс 
  pio run -e esp8266 -t upload прошивка
  pio run -e esp8266 -t uploadfs прошивка Фс
  git switch master переключение между ветками
  git switch feat переключение между ветками

dina16121974, 2025-05-24:
PS F:\FireLamp_JeeUI-master\FireLamp_JeeUI-master> pio pkg update
Resolving esp32 dependencies...
Already up-to-date.
Resolving esp32debug dependencies...
Resolving esp32S3 dependencies...
Already up-to-date.
Resolving hdwf2 dependencies...
Already up-to-date.
Resolving esp32_4M dependencies...
Already up-to-date.
Resolving esp32_16M dependencies...
Already up-to-date.
Resolving esp32S3_4M dependencies...
Already up-to-date.
Resolving esp32S3_16M dependencies...
Already up-to-date.
Resolving hdwf2_rel dependencies...
Already up-to-date.
PS F:\FireLamp_JeeUI-master\FireLamp_JeeUI-master> pio run -e esp32S3 -t upload
Processing esp32S3 (board: lolin_s3; framework: arduino; platform: https://github.com/tasmota/platform-espressif32/releases/download/2025.05.40/platform-espressif32.zip)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------Library Manager: Installing vortigont/EmbUI @ ~4.2.2
UnknownPackageError: Could not find the package with 'vortigont/EmbUI @ ~4.2.2' requirements for your system 'windows_amd64'
PS F:\FireLamp_JeeUI-master\FireLamp_JeeUI-master>

dina16121974, 2025-05-25:
скачал [Rel v4.1.1](https://github.com/vortigont/FireLamp_JeeUI/releases/tag/v4.1.1)
все нормально раскладывает
Resolving esp32S3 dependencies...
Library Manager: Installing mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display @ ~3.0
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESP32 HUB75 LED MATRIX PANEL DMA Display@3.0.12 has been installed!
Library Manager: Installing madhephaestus/ESP32Encoder @ ~0.11
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESP32Encoder@0.11.7 has been installed!
Library Manager: Installing olikraus/U8g2 @ ^2.34.17
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: U8g2@2.36.5 has been installed!
Library Manager: Installing git+https://github.com/finitespace/BME280
git version 2.41.0.windows.1
Cloning into 'C:\Users\DENIS\.platformio\.cache\tmp\pkg-installing-6ufs8yu8'...

Library Manager: BME280@3.0.0+sha.72a11d3 has been installed!
Library Manager: Installing vortigont/EmbUI @ ~4.2.1
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: EmbUI@4.2.1 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing bblanchon/ArduinoJson @ ^7.2,<7.4
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ArduinoJson@7.3.1 has been installed!
Library Manager: Installing esp32async/ESPAsyncWebServer @ ^3.6
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESPAsyncWebServer@3.7.4 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing ESP32Async/AsyncTCP @ ^3.3.8
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: AsyncTCP@3.3.8 has been installed!
Library Manager: Installing git+https://github.com/bblanchon/ArduinoStreamUtils
git version 2.41.0.windows.1
Cloning into 'C:\Users\DENIS\.platformio\.cache\tmp\pkg-installing-a3adervw'...

Library Manager: StreamUtils@1.9.0+sha.f9a88c3 has been installed!
Library Manager: Installing marvinroger/AsyncMqttClient @ ~0.9
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: AsyncMqttClient@0.9.0 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing arkhipenko/TaskScheduler @ ~3.7
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: TaskScheduler@3.7.0 has been installed!
Library Manager: Installing git+https://github.com/vortigont/FTPClientServer#feat
git version 2.41.0.windows.1
Cloning into 'C:\Users\DENIS\.platformio\.cache\tmp\pkg-installing-qh7t43fh'...

Library Manager: FTPClientServer@0.9.7+sha.a6e8a6e has been installed!
Library Manager: Installing vortigont/esp32-flashz @ ~1.1
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: esp32-flashz@1.1.5 has been installed!
Library Manager: Installing git+https://github.com/vortigont/LedFB
git version 2.41.0.windows.1
Cloning into 'C:\Users\DENIS\.platformio\.cache\tmp\pkg-installing-mzatffm7'...

Library Manager: LedFB@1.2.0+sha.ce35e5d has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing FastLED/FastLED @ 3.7.8
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: FastLED@3.7.8 has been installed!
Library Manager: Installing git+https://github.com/vortigont/TM1637
git version 2.41.0.windows.1
Cloning into 'C:\Users\DENIS\.platformio\.cache\tmp\pkg-installing-3s0xn_nu'...

Library Manager: TM1637 Driver@2.1.2+sha.666d348 has been installed!
Library Manager: Installing vortigont/ESPAsyncButton @ ~1.2
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESPAsyncButton@1.2.2 has been installed!
Library Manager: Installing git+https://github.com/vortigont/DFMiniMp3#notpls
git version 2.41.0.windows.1
Cloning into 'C:\Users\DENIS\.platformio\.cache\tmp\pkg-installing-085ae9cr'...

Library Manager: DFMiniMp3@1.2.3+sha.50d2b8f has been installed!
Library Manager: Installing git+https://github.com/vortigont/CronoS
git version 2.41.0.windows.1
Cloning into 'C:\Users\DENIS\.platformio\.cache\tmp\pkg-installing-4ktx26wl'...

Library Manager: CronoS@1.0.0+sha.83cb460 has been installed!
Updating metadata for the vscode IDE...
Project has been successfully updated!

closed 2025-09-28 by vortigont